adafruit/Adafruit_CircuitPython_BME280 #14: bme280.sea_level_pressure

grok00, 2018-09-12:
In example "bme280_simpletest.py" I see bme280.sea_level_pressure being set in the example but I don't see it being used.  This is an important feature for air pressure and elevation.  I can set this variable to any number and it makes no difference in the results.

Frank

caternuson, 2018-09-12:
It gets used in the altitude calculation here:
https://github.com/adafruit/Adafruit_CircuitPython_BME280/blob/master/adafruit_bme280.py#L184

Can you provide more detail of how you are setting it but not seeing any effect. A code snippet that demonstrates the issue would help.

grok00, 2018-09-12:
Caternuson,

I guess I expected to see a difference in the Pressure results but in fact it makes a big difference in the Altitude setting which actually makes the most sense (my Pressure readings still don't match the local weather service pressure results but I'm not sure how important that is and something I'll look at closer).

My mistake I didn't see Altitude change.  I guess I need to look closer at the code to see how the bme280.sea_level_pressure variable is picked up and used.  

Part of my learning experience with Circuit Python I guess!

I'm sorry I wasted your time and thank you for such a quick response!
Frank