adafruit/Adafruit_CircuitPython_BME280 #15: [Errno 121] Remote I/O error when reading temperature

jamesshannon, 2018-10-02:
I'm getting a remote I/O error when _read_temperature() is called. 

The initialization (which checks the chipid) works, and I'm able to do a few other things (like read the coefficients). Also, similar code has worked from the same environment with your Si7021 breakout.

```
root@monitoring_server:/# python
Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170118] on linux

>>> import board
>>> import busio
>>> import adafruit_bme280
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)
>>> bme280.humidity
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/adafruit_bme280.py", line 161, in humidity
    self._read_temperature()
  File "/usr/local/lib/python3.5/dist-packages/adafruit_bme280.py", line 93, in _read_temperature
    while self._read_byte(_BME280_REGISTER_STATUS) & 0x08:
  File "/usr/local/lib/python3.5/dist-packages/adafruit_bme280.py", line 218, in _read_byte
    return self._read_register(register, 1)[0]
  File "/usr/local/lib/python3.5/dist-packages/adafruit_bme280.py", line 243, in _read_register
    i2c.write(bytes([register & 0xFF]))
  File "/usr/local/lib/python3.5/dist-packages/adafruit_bus_device/i2c_device.py", line 102, in write
    self.i2c.writeto(self.device_address, buf, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/busio.py", line 63, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_blinka/microcontroller/raspi_23/i2c.py", line 38, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_PureIO/smbus.py", line 244, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error
```

jamesshannon, 2018-10-02:
Oddly, it works when I deconstruct the code and run individual commands:

```
>>> _write_register_byte(_BME280_REGISTER_CTRL_MEAS, 0xFE)
>>> _read_register(_BME280_REGISTER_STATUS, 1)
bytearray(b'\x00')
```

Might be a timing issue, but I tried putting both lines into a single function and running them together and that worked.

barbudor, 2019-06-01:
Not reproduced on Raspi rev 1, raspian-stretch, Adafruit-Blinka (1.2.8), adafruit-circuitpython-busdevice (2.2.11)

jamesshannon, 2019-06-06:
No. I haven't run into this in a while. Don't remember if it fixed itself or what.

closed 2019-06-06 by jamesshannon