adafruit/Adafruit_CircuitPython_BME280 #21: needs oversampling + IIR Filter options

jonas87, 2019-02-12:
If I use this library and keep a log of the measurements, the temperature and humidity data are super noisy.
I tried different measurements intervals in the range [2;120] seconds, but all of it is noisy. The pressure data on the other hand is smooth. 

The BME280 has support for supersampling and has a built in IIR filter, both of which help smoothing out the data. Currently the library only has supersampling enabled for humidity (hardcoded), it would be useful if the value can be set, and also for temperature and pressure. The IIR filter currently does not seem to be used?

[image]

ladyada, 2019-02-12:
correct! if you'd like to try enabling that oversampling support and submitting a PR we'd be psyched to add it :)

makermelissa, 2019-03-13:
Fixed with #22.

closed 2019-03-13 by makermelissa