adafruit/Adafruit_CircuitPython_BME280 #25: Don't use frozenset

ladyada, 2019-05-21:
not included on small CP boards :)

tannewt, 2019-05-31:
We should replace them with tuples.

loganwedwards, 2019-06-03:
Is this related to not being able to use the BME 280 with the Feather M0 RFM9x. I am using the provided sample code from the tutorials and this is where it fails.
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 5, in <module>
  File "adafruit_bme280.py", line 77, in <module>
NameError: name 'frozenset' is not defined
```

loganwedwards, 2019-06-03:
With further research, it appears PR #26 will address this.

commit 53e09304 (2019-06-17): Merge pull request #26 from barbudor/master

fix #25 + fix #19 + refactor `pressure` property
closed 2019-06-17 by ladyada