adafruit/Adafruit_CircuitPython_BME280 #31: [Errno 121] Reoccur

nmaas87, 2019-12-30:
Dear all,

I use the Adafruit CircuitPython Ecosystem now with a whole strip of sensors:
1x BME280, 1x LSM9DS1, 2x INA219 (one  as 0x40, one as 0x41) and a DS3231 - which is currently not actively pulled. I actively try to get data from all sensors as fast as possible and save this info in a dict.
All is run on a Raspberry Pi Zero W, with current drivers and dependencies.

````
Adafruit-Blinka==3.3.4
adafruit-circuitpython-bme280==2.3.2
adafruit-circuitpython-busdevice==4.1.1
adafruit-circuitpython-ds3231==2.1.3
adafruit-circuitpython-ina219==3.3.0
adafruit-circuitpython-lsm9ds1==2.0.5
adafruit-circuitpython-register==1.7.1
Adafruit-PlatformDetect==1.3.8
Adafruit-PureIO==1.0.4
pyftdi==0.42.2
pyusb==1.0.2
rpi-ws281x==4.2.2
spidev==3.4
sysv-ipc==1.0.1
````

After some time, the whole script "dies", always pointing to the BME280 lib, which seems to be similiar to #15 

````
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "sensorTest.py", line 41, in gpioThreadFunc
    tlm["bt"], tlm["bp"], tlm["ba"], tlm["bh"]  = bme280.temperature, bme280.humidity, bme280.pressure, bme280.altitude
  File "/usr/lib/python3.7/site-packages/adafruit_bme280.py", line 335, in pressure
    self._read_temperature()
  File "/usr/lib/python3.7/site-packages/adafruit_bme280.py", line 143, in _read_temperature
    while self._get_status() & 0x08:
  File "/usr/lib/python3.7/site-packages/adafruit_bme280.py", line 172, in _get_status
    return self._read_byte(_BME280_REGISTER_STATUS)
  File "/usr/lib/python3.7/site-packages/adafruit_bme280.py", line 426, in _read_byte
    return self._read_register(register, 1)[0]
  File "/usr/lib/python3.7/site-packages/adafruit_bme280.py", line 451, in _read_register
    i2c.write(bytes([register & 0xFF]))
  File "/usr/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py", line 98, in write
    self.i2c.writeto(self.device_address, buf, **kwargs)
  File "/usr/lib/python3.7/site-packages/busio.py", line 84, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/usr/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 38, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/usr/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 256, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error
````

Test Code:
````
from time import sleep, time
import threading
import board
import busio
import adafruit_bme280
import adafruit_lsm9ds1
from adafruit_ina219 import ADCResolution, BusVoltageRange, INA219
#import adafruit_ds3231
import json
tlm = {}

# Create library object using our Bus I2C port
i2c = busio.I2C(board.SCL, board.SDA, frequency=400000)
bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)
lsm9ds1 = adafruit_lsm9ds1.LSM9DS1_I2C(i2c)
ina219 = INA219(i2c)
ina219b = INA219(i2c, 0x41)

def i2c_writeNumber(value):
	value = chr(value)
	i2c.writeto(0x26,value)

# change this to match the location's pressure (hPa) at sea level
bme280.sea_level_pressure = 1013.25

# optional : change configuration to use 32 samples averaging for both bus voltage and shunt voltage
ina219.bus_adc_resolution = ADCResolution.ADCRES_12BIT_32S
ina219.shunt_adc_resolution = ADCResolution.ADCRES_12BIT_32S
# optional : change voltage range to 16V
ina219.bus_voltage_range = BusVoltageRange.RANGE_16V

# optional : change configuration to use 32 samples averaging for both bus voltage and shunt voltage
ina219b.bus_adc_resolution = ADCResolution.ADCRES_12BIT_32S
ina219b.shunt_adc_resolution = ADCResolution.ADCRES_12BIT_32S
# optional : change voltage range to 16V
ina219b.bus_voltage_range = BusVoltageRange.RANGE_16V

def gpioThreadFunc():
	while True:
		#print("BME280")
		tlm["bt"], tlm["bp"], tlm["ba"], tlm["bh"]  = bme280.temperature, bme280.humidity, bme280.pressure, bme280.altitude
		
		sleep(0.01)
		#print("LSM9DS1")
		# Read acceleration, magnetometer, gyroscope, temperature.
		tlm["ax"], tlm["ay"], tlm["az"] = lsm9ds1.acceleration
		tlm["mx"], tlm["my"], tlm["mz"] = lsm9ds1.magnetic
		tlm["gx"], tlm["gy"], tlm["gz"] = lsm9ds1.gyro

		sleep(0.01)
		tlm["5b"], tlm["5s"], tlm["5c"] = ina219.bus_voltage, ina219.shunt_voltage, ina219.current

		sleep(0.01)
		tlm["3b"], tlm["3s"], tlm["3c"] = ina219b.bus_voltage, ina219b.shunt_voltage, ina219b.current

		sleep(0.01)		
		print(tlm)
		i2c_writeNumber(1)
		#sleep(0.05)
		#sleep(0.75)

def mainThreadFunc():
	sleep(5)

gpioThread = threading.Thread(target = gpioThreadFunc)
mainThread = threading.Thread(target = mainThreadFunc)
gpioThread.start()
mainThread.start()

````

The main thread is so work with the data and transform it, but I have not yet gotten so far.
Do you see anything that I am doing wrong - and if so, how can I fix it / are there besides this problems ways to enhance speed / more readings per second :)?

The only special thing I added, was to sent one byte to the i2c Device on 0x26. But other than that, it is all fairly standard.

Thanks a lot for your help and awesome work over at Adafruit!

Nico

ladyada, 2019-12-30:
try without threads, make sure your wiring is solid, and you can always re-try-except the sensor reads

ladyada, 2020-03-10:
closing due to no response

closed 2020-03-10 by ladyada