adafruit/Adafruit_CircuitPython_BME280 #39: Documentation

crablab, 2020-06-27:
Hey :wave: 

I note that this is missing Pydoc. 

I'm happy to go through and add some if that would be useful? :)

ladyada, 2020-06-28:
@tannewt hihi you're the documentation expert, whatcha think? :)

tannewt, 2020-06-29:
@crablab What do you mean by Pydoc? It looks like the properties are documented: https://github.com/adafruit/Adafruit_CircuitPython_BME280/blob/master/adafruit_bme280.py#L295-L298

A PR to improve docs is always welcome!

crablab, 2020-06-29:
Yeah, basically that but there are some places where it is missing and no explicit params or return values are supplied :) 

eg. https://github.com/adafruit/Adafruit_CircuitPython_BME280/blob/394cb7aa225e8c5b6ede77067aa82f4017f0d814/adafruit_bme280.py#L479

If contributions to docs are okay, maybe I can take a look through and update some of the comments and then PR it? You can then yay or nay it :blush:

tannewt, 2020-06-30:
@crablab Yup! All kinds of contributions are encouraged. Let me know if you need any help getting going. Most of the devs are available for questions on the Adafruit Discord too: https://adafru.it/discord

dherrada, 2020-08-06:
@crablab Hey, is this something you're still interested in doing? If not, that's totally fine

crablab, 2020-08-07:
Hey! Sorry, I missed this :see_no_evil:  I'll have a look this weekend if that's okay?

dherrada, 2020-08-07:
@crablab No worries. This pr would be greatly appreciated, but don't feel obligated to work on it.

crablab, 2020-08-07:
@dherrada For sure :smile: I'll update this at the weekend with what I have/haven't done to give you a steer! :smile:

jposada202020, 2021-04-08:
@crablab Hello, would you like to work on this 😄  I can take a look during ``this`` weekend if you want

crablab, 2021-04-19:
@jposada202020 Feel free! :)

dherrada, 2021-04-19:
closed by #46

closed 2021-04-19 by dherrada