adafruit/Adafruit_CircuitPython_BME280 #44: Support out of full accuracy pressure

gbaman, 2021-04-06:
In the datasheet for the BME280, it states the pressure range of the sensor is full accuracy 300hPa to 1100hPa, which no doubt is perfectly fine for just about everyone.
Not noticing this on the datasheet, we launched 2 BME280s on a high altitude weather balloon. The first used this library and the second used a different library.
The sensor itself quite happily returned results right down to 50hPa on the second sensor (after which the balloon burst), but the first sensor that used this library simply returned 300 at that point.

If the sensor is capable of returning values (even if outside the "full accuracy" range), would it not make sense to simply return these values instead of artificially capping it?

Temperature is simply returned no matter the value, even if it is only rated down to -40c. Funny enough, we reported down to -46c on this flight without issues!

[image]

ladyada, 2021-04-06:
ok wanna try a PR removing
https://github.com/adafruit/Adafruit_CircuitPython_BME280/blob/master/adafruit_bme280.py#L371
?

gbaman, 2021-04-06:
> 
> 
> ok wanna try a PR removing
> https://github.com/adafruit/Adafruit_CircuitPython_BME280/blob/master/adafruit_bme280.py#L371
> ?

Done https://github.com/adafruit/Adafruit_CircuitPython_BME280/pull/45

closed 2021-04-06 by gbaman