adafruit/Adafruit_CircuitPython_BME280 #53: Did something with this project recently change? The examples no longer work.

xekon, 2021-06-10:
I used the examples here as a starting point for a project on a raspberry pi over a year ago, it has worked without issue.

https://github.com/adafruit/Adafruit_CircuitPython_BME280/blob/main/examples/bme280_simpletest.py

I recently tried to recreate the project on another raspberry pi but this library seems to have changed?

I installed it like this:

pip3 install smbus2 adafruit-circuitpython-bme280

before I had these lines:
```
import adafruit_bme280
i2c = busio.I2C(board.SCL, board.SDA)
bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c, 0x76)
```

now I have:
```
import Adafruit_BME280
i2c = busio.I2C(board.SCL, board.SDA)
bme280 = Adafruit_BME280.Adafruit_BME280_I2C(i2c, 0x76)
```

Notice I had to capitalize the library, which is fine however I am now seeing: 

`AttributeError: module 'Adafruit_BME280' has no attribute 'Adafruit_BME280_I2C'`

additionally the guide on the main page for this project appears to be out of date as well: https://github.com/adafruit/Adafruit_CircuitPython_BME280#usage-example

Really appreciate any help on this, Thanks a Ton!

xekon, 2021-06-10:
Seems the copy here on github is fine, it is just the copy in the pip3 repository that is not. I worked around the problem like this:

where ever you main python project file is change to that directory, mine is under root at /root/sensor.py

`wget https://raw.githubusercontent.com/adafruit/Adafruit_CircuitPython_BME280/2.6.5/adafruit_bme280/basic.py`

nano sensor.py  then instead of
`import Adafruit_BME280`

do this which will import the file you just downloaded:
`import basic`

then farther below:
`bme280 = basic.Adafruit_BME280_I2C(i2c, 0x76)`

jposada202020, 2021-06-10:
Thank you, yes this library have changed with the new release.
I will clarify the `readme.rst`, and usage inside the library to be the same as in the `bme280_simpletest.py`

xekon, 2021-06-10:
Thanks, there is a TON of guides on the net for the BME280 sensor that reference the old code from before the new library release. 

The one I downloaded via PIP3 did not working according to those guides so when the readme and examples on this github page also did not work it was just a little confusing at first.

jposada202020, 2021-06-10:
No problem, hope is more clear now, I added some notes in the readme. And thank you for pointing this out :)
for future reference we have the learning guide in 
https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/python-circuitpython-test
and we try to keep it up to date when we introduce new changes

xekon, 2021-06-10:
Thanks, there is a TON of guides on the net for the BME280 sensor that reference the old code from before the new library release.

The one I downloaded via PIP3 did not working according to those guides so when the readme and examples on this github page also did not work it was just a little confusing at first.

The main readme page (https://github.com/adafruit/Adafruit_CircuitPython_BME280#installing-from-pypi) now show to install like this:

`pip3 install adafruit-circuitpython-bme280`

however when you do that, you cannot do this (as per the readme https://github.com/adafruit/Adafruit_CircuitPython_BME280#usage-example) :

`from adafruit_bme280 import basic`

it does not work

```
~# ./sensor.py
Traceback (most recent call last):
  File "./sensor.py", line 6, in <module>
    from adafruit_bme280 import basic
ModuleNotFoundError: No module named 'adafruit_bme280'
```

you can however do this:

`import Adafruit_BME280`

but you cannot do:

`from Adafruit_BME280 import basic`

```
~# ./sensor.py
Traceback (most recent call last):
  File "./sensor.py", line 6, in <module>
    from Adafruit_BME280 import basic
ImportError: cannot import name 'basic' from 'Adafruit_BME280' (/usr/local/lib/python3.7/dist-packages/Adafruit_BME280/__init__.py)
```
Notice the one installed via pip3 is "Adafruit_BME280" not adafruit_bme280" and also the capitalized one installed via pip3 does not seem to have the basic module available.

the only solution that worked for me is to NOT use PIP3 to install and instead download the file directly from this github.

jposada202020, 2021-06-10:
You are right,  have just changed the way to upload the library to pipy for this to be treated as a package. These change needs to be merged and release, before we will see the change

xekon, 2021-06-10:
Ah that is great news, so it will get fixed eventually, thanks for your quick response.

closed 2021-06-10 by xekon

nathanbent, 2022-05-10:
Almost a year later and the same problem still exists, is there any chance of getting the version installed by pip3 updated?

Ozzyben, 2022-07-22:
Current version still gives same error when using the example code from pypi:
```
Traceback (most recent call last):
  File "/home/pi/sensor.py", line 7, in <module>
    bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)
AttributeError: module 'adafruit_bme280.basic' has no attribute 'Adafruit_BME280_I2C'
```

PR #54 (2021-06-10): aligning_example_code and adding_library_usage_explanation, updating PyPI_packaging_Info
Aligning example code and adding library usage explanation in the `readme.rst`

see #53 

This will also change `py_modules` to `packages` in the `conf.py` to resolve #53 in PyPI