adafruit/Adafruit_CircuitPython_BME280 #59: FTDI on Linux PC

zenon2cubed, 2022-01-30:
I have a genuine FTDI dongle and Ubuntu PC.  Whenever the "import board" command is run I get the following error:

`>>> import board`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`  File "/usr/local/lib/python3.8/dist-packages/board.py", line 294, in <module>`
`    raise NotImplementedError("Board not supported {}".format(board_id))`
`NotImplementedError: Board not supported GENERIC_LINUX_PC`

Based on the online tutorial I was not expecting a problem with support for generic linux PC's.  Please advise.  thanks.

caternuson, 2022-01-30:
Did you set the environment variable before launching the Python session?
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting#check-environment-variable-within-python-3041702-11

zenon2cubed, 2022-01-30:
I think so.  See output below:

`>>> import os`

`>>> os.environ["BLINKA_FT232H"]`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`  File "/usr/lib/python3.8/os.py", line 675, in __getitem__`
`    raise KeyError(key) from None`
`KeyError: 'BLINKA_FT232H'`

`>>> import board`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`  File "/usr/local/lib/python3.8/dist-packages/board.py", line 294, in <module>`
`    raise NotImplementedError("Board not supported {}".format(board_id))`
`NotImplementedError: Board not supported GENERIC_LINUX_PC`

caternuson, 2022-01-30:
[image]

zenon2cubed, 2022-01-30:
What did I do wrong in setting the environment variable?  How do I know what the other options are to decide the correct setting?

ladyada, 2022-01-30:
hiya, questions about how to use adafruit products are best posted in the adafruit forums  at [forums.adafruit.com](forums.adafruit.com/)
thats where tech support is handled - github issues are where folks can issue bug reports (e.g. something that ought to work but doesnt due to a hardware or software bug)

thanks! :)

closed 2022-01-30 by ladyada