adafruit/Adafruit_CircuitPython_BME280 #61: duplicate-code check is not happy

evaherrada, 2022-06-21:
https://github.com/adafruit/Adafruit_CircuitPython_BME280/runs/6993430685?check_suite_focus=true#step:11:33

kattni, 2022-06-21:
Assigning Tekktrik as well to look into this. We'll work together to decide what the best solution is.

tekktrik, 2022-06-21:
Interesting that it thinks it's duplicate code even though one is I2CDevice and the other is SPIDevice.  This could also be refactored, but see more of an argument for ignoring it given it's not *truly* duplicate.

tekktrik, 2022-06-21:
Oh, never mind, I read the CI wrong.  It's likely we could set up a base class for each device type.

rsbohn, 2022-08-19:
Going to give the refactor a try. #CircuitPythonDay2022

rsbohn, 2022-08-20:
I've got a pull request: https://github.com/adafruit/Adafruit_CircuitPython_BME280/pull/62

Questions about naming things: 

I extracted the bus specific parts to classes named I2C_Impl and SPI_Impl in protocol.py. Are these *_Impls, proxies, or delegates? Is there a better name than protocol.py?

rsbohn, 2022-08-20:
Tested with Blinka/MCP2221.
Tested with 8.0.0-beta.0 on 2022-08-18; Adafruit QT Py M0 with samd21e18:
Had to run the .mpy; no room for the advanced.py or .mpy files.  But it worked!

```import board
from adafruit_bme280 import basic

i2c = board.I2C()
sensor = basic.Adafruit_BME280_I2C(i2c, address=0x76)
print("Temp:",sensor.temperature)
print("RH:  ",sensor.relative_humidity)
print("Alt: ",sensor.altitude)
```

closed 2022-09-26 by FoamyGuy