adafruit/Adafruit_CircuitPython_BME280 #66: issue

Lbenhabiles, 2023-02-11:
Hello, 

I am having this issue, any idea please ? 

Traceback (most recent call last):
  File "/home/pi/n.py", line 16, in <module>
    bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bme280/advanced.py", line 321, in __init__
    super().__init__(I2C_Impl(i2c, address))
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bme280/advanced.py", line 152, in __init__
    super().__init__(proxy)
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bme280/basic.py", line 107, in __init__
    self._read_coefficients()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bme280/basic.py", line 292, in _read_coefficients
    self._humidity_calib[0] = self._read_byte(0xA1)  # BME280_REGISTER_DIG_H1
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bme280/basic.py", line 303, in _read_byte
    return self._read_register(register, 1)[0]
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bme280/basic.py", line 314, in _read_register
    return self._bus_implementation.read_register(register, length)
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bme280/protocol.py", line 23, in read_register
    i2c.write(bytes([register & 0xFF]))
  File "/usr/local/lib/python3.9/dist-packages/adafruit_bus_device/i2c_device.py", line 101, in write
    self.i2c.writeto(self.device_address, buf, start=start, end=end)
  File "/usr/local/lib/python3.9/dist-packages/busio.py", line 174, in writeto
    return self._i2c.writeto(address, memoryview(buffer)[start:end], stop=stop)
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 52, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_PureIO/smbus.py", line 314, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

caternuson, 2023-02-11:
Looks like a connection issue. Please post in the forums with photos of your setup showing how everything is connected:
https://forums.adafruit.com/

ladyada, 2023-03-15:
no reply

closed 2023-03-15 by ladyada