adafruit/Adafruit_CircuitPython_BME280 #70: Broken package?

sigmapi332, 2025-04-01:
Is this package broken?  I've tried installing it at least time times with uninstalls in-between. It says it installs fine but then there's effectively nothing actually installed.  Looking at the __init__.py file in both the files I installed using PIP and via the files I hauled down using GIT, they are all blank.

caternuson, 2025-04-01:
What platform are you using? Raspberry Pi? Something else?

Are you using a virtual environment?

I just tested a pip install on a Raspberry Pi and it seemed to work OK.

```
(env) pi@raspberrypi:~/env/lib/python3.11/site-packages/adafruit_bme280$ ls -al
total 44
drwxr-xr-x  3 pi pi  4096 Apr  1 15:21 .
drwxr-xr-x 39 pi pi  4096 Apr  1 15:21 ..
-rw-r--r--  1 pi pi 11703 Apr  1 15:21 advanced.py
-rw-r--r--  1 pi pi 14656 Apr  1 15:21 basic.py
-rw-r--r--  1 pi pi     0 Apr  1 15:21 __init__.py
-rw-r--r--  1 pi pi  2139 Apr  1 15:21 protocol.py
drwxr-xr-x  2 pi pi  4096 Apr  1 15:21 __pycache__
```
`env` is the name of the virtual environment set up on that pi.

sigmapi332, 2025-04-01:
I’ve tried it both with Bullseye and in a virtual environment on Bookworm with the same result unfortunately. 

It’s weird that when I view __init__.py on the repository it is blank as well.

caternuson, 2025-04-01:
What is the full pip command you are using to install the package into the venv?

sigmapi332, 2025-04-02:
I've tried:
pip3 install adafruit-circuitpython-bme280
sudo pip3 install adafruit-circuitpython-bme280

and for good measure

pip install adafruit-circuitpython-bme280
sudo pip install adafruit-circuitpython-bme280

caternuson, 2025-04-02:
Don't use sudo. Please read through this guide:
https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi/overview
and make sure this isn't just a sudo/venv issue.

caternuson, 2025-04-15:
Closing due to lack of response. Could not recreate locally. Most likely related to sudo/venv behavior.